MiForge/MiUnlockTool #7: MacOS Error

Jonnes, 2024-03-05:
terminal:

```
invalid product or deviceToken

Check if device is connected in bootloader mode...

product saved.
```

and the R S OK page url doesn't return deviceToken

Jonnes, 2024-03-05:
it is only for global edition?
i use Chinese edition and i see the cookies records about the page. only serviceToken no token or deviceToken

Jonnes, 2024-03-05:
can't get any token. after execute the script, the token of data.json is still empty. I check the link of the page and the cookies, it seems the server will not return the value of token or deviceToken, only serviceToken

Jonnes, 2024-03-05:
so when i paste the link to the terminal, it shows `invalid product or token`

Jonnes, 2024-03-05:
and it won't abort. just keep looping this:
`invalid product or token

Check if device is connected in bootloader mode...

product saved.`

Jonnes, 2024-03-05:
yes, already

Jonnes, 2024-03-05:
`

invalid product or token

Check if device is connected in bootloader mode...

token saved.

Check if device is connected in bootloader mode...

product saved.

product: zircon

deviceToken: 

Account Region: china

`

Jonnes, 2024-03-06:
it is not the same way to get token from a Qualcomm or Mediatek.
Mediatek device uses this:
fastboot oem get_token

Jonnes, 2024-03-06:
`fastboot getvar token` is for Snapdragon devices. Therefore the script gets nothing and loop.

Jonnes, 2024-03-06:
yes, it is finally fastboot oem get_token

Jonnes, 2024-03-06:
```
Last login: Wed Mar  6 20:12:35 on ttys001
jonnes@192 platform-tools % ./fastboot getvar all
(bootloader) 	build_adc: 136
(bootloader) 	project_adc: 1438
(bootloader) 	hwversion: 3.9.0
(bootloader) 	hwlevel: MP
(bootloader) 	hwc: CN
(bootloader) 	off-mode-charge: 1
(bootloader) 	slot-retry-count:b: 1
(bootloader) 	slot-retry-count:a: 1
(bootloader) 	slot-successful:b: yes
(bootloader) 	slot-successful:a: yes
(bootloader) 	slot-unbootable:b: no
(bootloader) 	slot-unbootable:a: no
(bootloader) 	slot-count: 2
(bootloader) 	current-slot: b
(bootloader) 	rpmblock_support: yes
(bootloader) 	tokenversion: 2
(bootloader) 	anti: 1
(bootloader) 	fuse: yes
(bootloader) 	unlocked: yes
(bootloader) 	secure: no
(bootloader) 	partition-type:userdata: f2fs
(bootloader) 	partition-size:userdata: 7408ff8000
(bootloader) 	has-slot:userdata: no
(bootloader) 	partition-type:vbmeta_vendor_b: raw
(bootloader) 	partition-type:vbmeta_vendor_a: raw
(bootloader) 	partition-size:vbmeta_vendor_b: 800000
(bootloader) 	partition-size:vbmeta_vendor_a: 800000
(bootloader) 	has-slot:vbmeta_vendor: yes
(bootloader) 	partition-type:vbmeta_system_b: raw
(bootloader) 	partition-type:vbmeta_system_a: raw
(bootloader) 	partition-size:vbmeta_system_b: 800000
(bootloader) 	partition-size:vbmeta_system_a: 800000
(bootloader) 	has-slot:vbmeta_system: yes
(bootloader) 	partition-type:vbmeta_b: raw
(bootloader) 	partition-type:vbmeta_a: raw
(bootloader) 	partition-size:vbmeta_b: 800000
(bootloader) 	partition-size:vbmeta_a: 800000
(bootloader) 	has-slot:vbmeta: yes
(bootloader) 	partition-type:super: raw
(bootloader) 	partition-size:super: 220000000
(bootloader) 	has-slot:super: no
(bootloader) 	partition-type:md1img_b: raw
(bootloader) 	partition-type:md1img_a: raw
(bootloader) 	partition-size:md1img_b: c800000
(bootloader) 	partition-size:md1img_a: c800000
(bootloader) 	has-slot:md1img: yes
(bootloader) 	partition-type:dtbo_b: raw
(bootloader) 	partition-type:dtbo_a: raw
(bootloader) 	partition-size:dtbo_b: 800000
(bootloader) 	partition-size:dtbo_a: 800000
(bootloader) 	has-slot:dtbo: yes
(bootloader) 	partition-type:vendor_boot_b: raw
(bootloader) 	partition-type:vendor_boot_a: raw
(bootloader) 	partition-size:vendor_boot_b: 4000000
(bootloader) 	partition-size:vendor_boot_a: 4000000
(bootloader) 	has-slot:vendor_boot: yes
(bootloader) 	partition-type:boot_b: raw
(bootloader) 	partition-type:boot_a: raw
(bootloader) 	partition-size:boot_b: 4000000
(bootloader) 	partition-size:boot_a: 4000000
(bootloader) 	has-slot:boot: yes
(bootloader) 	partition-type:lk_b: raw
(bootloader) 	partition-type:lk_a: raw
(bootloader) 	partition-size:lk_b: 1000000
(bootloader) 	partition-size:lk_a: 1000000
(bootloader) 	has-slot:lk: yes
(bootloader) 	version-baseband: MOLY.NR16.R1.MP3P5.TC8.PR1.SP.V2.P14.CN
(bootloader) 	protocol_version: 1
(bootloader) 	is-userspace: no
(bootloader) 	serialno: 8DLBUOTCQWXORGCM
(bootloader) 	product: zircon
(bootloader) 	max-download-size: 0x8000000
(bootloader) 	version-bootloader: zircon_7391e8610_202312291129
(bootloader) 	version: 0.5
(bootloader) 	crc: 1
all: Done!!
Finished. Total time: 0.021s
```

tux-86, 2024-03-07:
I have also zircon, and when I run this script on linux, I get following variables but the script running in loop.

invalid product or token 

DeviceInfo:
product: zircon
token: VQECOAEQPra2GNtiPqOt+JUTl3uyNwMGemlyY29u

AccountInfo:
id: hidden
region: europe

tux-86, 2024-03-07:
With command "fastboot oem get_token" I get two token lines:
(bootloader) token: VQECOAEQPra2GNtiPqOt+JUTl3uyNwMGemlyY29u
(bootloader) token: AhyYcak2syPa42GMTYiFp5FxIsbeBnFArhSPwum3

tux-86, 2024-03-07:
I think that merged token is correct one, because the script now ask "Please add your account in Settings > Developer options > Mi Unlock status".

macekboris, 2024-05-14:
4 MTK 👍 
ADB shell 👍 
```
@Echo off
Rem StratOS@xda
adb kill-server
adb reboot bootloader
Echo press key when entered the FASTBOOT
pause
CLS
fastboot devices
fastboot oem get_token 2>&1 | findstr /l /b /c:"(bootloader) token:">token.bin
fastboot reboot
```
token.bin
`(bootloader) token: VQECOAEQZsg74/1Vyklpe4FUCuP0mAMGemlyY29u
(bootloader) token: Ahyey+EIG86fz71U1h1wL1uUFbbepfkuz2DJ4/B9`

You must join the string 👍 
`VQECOAEQZsg74/1Vyklpe4FUCuP0mAMGemlyY29uAhyey+EIG86fz71U1h1wL1uUFbbepfkuz2DJ4/B9`