MiForge/MiUnlockTool #15: Code 20038

gmvlasta, 2024-04-01:
I am trying to unlock the Redmi 5 Plus (vince). I finished with this:

=========================================================
product saved successfully
token saved successfully

DeviceInfo:
product: vince
token: ******

AccountInfo:
id: ******
region: *******
______________________________________________________

code: 20038

description: 该手机已被账号 145****753 通过查找手机锁定，无法解锁
==========================================================

What I found strange is that in settings there is information that the device is unlocked. But the output of command "fastboot oem device-info" is different:

(bootloader) 	Device unlocked: false
(bootloader) 	Device critical unlocked: false

I would appreciate any idea how to solve it. Thank you.

gmvlasta, 2024-04-03:
Thank you for help. I also think that the device is probably linked to
another account.

Settings in device was: 
Developer options » Mi unlock status » This device is unlocked

I tried your steps:

fastboot getvar unlocked 
RESULT
unlocked: no

fastboot oem lock
RESULT
(bootloader) Device already : locked!

Then I did factory reset but the state in settings is the same:

Developer options » Mi unlock status » This device is unlocked

Maybe it could help to flash the MIUI firmware via fastboot if possible?

gmvlasta, 2024-05-15:
As the device was new, I claimed a warranty and a seller returned me
my money. So I couldn't try your last advices. I believe that it would
solve the issue. Thank you very much for help!

bodV6, 2024-09-24:
On line 37,
How can I change the state of:
ro.secureboot.lockstate: locked

So that mi status unlock can open to add my account again.
Or any other advice to unlock bootloader.

Mi assistant tool only flashed OTA on recovery mode, but did not change the lock state.

offici5l, 2024-09-26:
If you tried flashing recovery ROM xith mi assistant and still have the same issue ..
So you should try other solutions...
Unfortunately, I have no idea at the moment...

slg407, 2024-11-20:
to add onto this, this error, specifically on brand new devices, back in the day was usually related to the reseller putting a malicious custom ROM on the device, if this happens to you check the product authentication page on the xiaomi website, and it will almost 100% show you that the device is not the official global version, this used to be a much larger problem with older xiaomi devices, where resellers would reflash the CN version of the device with a version of MIUI packed with malware and then resell the device for cheap, profiting off of stolen data afterwards

bodV6, 2024-11-20:
Thanks for this info.
Fortunately for me, where while desperately looking for unlock solutions. I have opened the back cover and expose the EDL panel of my phone because Redmi  k40 pro only works through EDL test point and not USB EDL.

I flashed the Official China ROM, and the bootloader is now evidently LOCKED. 
I was thinking from there I can add my mi account and unlock.
Unfortunately, new Error now from mi unlock is:
 "THE PLACE WHERE THE ACCOUNT IS REGISTERED DOESN'T MATCH THE PLACE WHERE THE PHONE IS SOLD".

It seems xiaomi does not want to unlock bootloader again. Futile efforts ever since.

Specifically bought EOL Device for custom roms.

bodV6, 2024-11-20:
Additionally, if I flash official Global or EU or any MIUI ROM.
It results to "System Destroyed" I had to occasionally revert to flash back the Official China HyperOS RoM.

slg407, 2024-11-20:
please send the results of: https://www.mi.com/global/verify/

also, if you managed to install the chinese ROM i imagine you have the patched firehose loader file?

also also, "the system is destroyed" could be related to anti-rollback, and not with the rom itself, one thing you could try is using https://github.com/bkerler/edl with the "edl modules oemunlock enable" command, by loading the patched firehose file you have (or use the tool's automatic detection to search if one of the pre-packaged loaders works with your device), i am not responsible for damages that could happen with this tool, but it could be useful for your quest if you are skilled enough to learn to use it, good luck

bodV6, 2024-11-21:
Thank you slg407, the Device Xiaomi Redmi 40 pro is verified original.
But you understand the Device was specifically made for China region, while it's global equivalent is called Xiaomi Mi 11i.(With global ROM)

Actually if I try to downgrade on EDL to lower Chinese ROM, I will get Anti Rollback Error. Unable flash error on mi Flash tool.

Only flashing Global ROM causes system Destroyed.(Successful flashed, but display system destroyed on boot, nothing else works)

I will try the tool you suggested and give a feedback here.
Thanks Again.

bodV6, 2024-11-22:
@slg407 I have been able to setup and py build the tool on windows, but errors with lib and USB does not make edl loader detect the phone. hope you can help out. but here is what I have tried,
I ran Zadig tool, but no devices appears, 
at a point I replaced libusb_win32 driver with the Q9008 com for edl. but reverted back since it did not work.
edl detects the phone at COM6, I only don't understand the need to detect USB details.

EDL for redmi k40 pro uses onboard edl point to short circuit. only Q9008 works

here is the error:
PS C:\Users\abode\Sahara Qualcomm Diag Tool\edl> edl --serial
main
main - [LIB]: Please first install libusb_win32 driver from Zadig
main - Trying with no loader given ...
main - Waiting for the device
Detected 0x5c6:0x9008 device at: COM6
main - Device detected :)
DeviceClass - Warning !
main - Mode detected: sahara
Traceback (most recent call last):
  File "C:\Users\abode\Sahara Qualcomm Diag Tool\edl\edl", line 393, in <module>
    base.run()
  File "C:\Users\abode\Sahara Qualcomm Diag Tool\edl\edl", line 293, in run
    version = conninfo.get("data").version
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

slg407, 2024-11-27:
inside the edl tool there's the drivers folder:
[image]
inside of it a folder named "Windows"
[image]
inside of it is all you need
[image]
extract the 7-zip, then plug in your phone while on EDL mode and run the zadig tool included
[image]
select the android device from the dropdown (you may need to open device manager to see it, and it may have a weird name)
then select the libusb-win32 driver on the options and click the big button
[image]

bodV6, 2024-11-27:
WinUSB works, to replace the Q9008.
I was able to do that yesterday.
But the new error is that none of the 
Firehose loader works for my devices.
The tool tried all firehose loader that matched my device hash.
But none worked.
It disconnected after trying the last.

bodV6, 2024-11-27:
Please can you help get the right Firehose loader for Redmi K40 pro

offici5l, 2025-04-07:
code: 20038
Message: 该手机已被账号 **** 通过查找手机锁定，无法解锁
Description: device locked by account ****
Solution: Log in to account **** in the following link: i.mi.com/mobile/find, select device, Disable Locate Device

closed 2025-04-07 by offici5l