MiForge/MiUnlockTool #34: Can't get token after logging in

Guybrush3pw00d, 2024-05-04:
My best regards. I'm stuck with the dreaded "device not activated" error (apart from the fact that activating hasn't ever been a compulsory step while just binding under developer options has always been enough, my device is correctly activated instead) arisen only after waiting for the time asked by Xiaomi and I've tired to install miunlock according to the guidelines.
I log in with Mi account only after launching the process and I start with a browser with empty cache. Tried three different browsers too (Kiwi, Fennec and Soul) with the same outcome. After logging in no confirmation page with the token in its address but just errors (401 with Kiwi and Fennec, net::ERR_HTTP_RESPONSE_CODE_FAILURE with Soul) at [this](https://unlock.update.miui.com/sts?d=wb_00b247f8-e4fe-485d-8a63-66b89cffce7f&ticket=0&pwd=1&p_ts=1714819455000&fid=0&p_lm=1&auth=YQkdtTX70BmdX1XcBFzTXdGpHb8%2Fz5Y0PLLrQzu1DkVu12%2FRolP%2FnElmWbcUeDnT%2FSOxYbHDMdVWDsqkuvbhjuqYAIGONihEVdLR%2Bu2xu73%2BS%2FDHTk8l3cgKHBH3tw5G14cMbIJw4%2BlOs%2Bv3vTTzo1Z6YQsLBGjSXYLUDlOTfzM%3D&m=9&_group=DEFAULT&tsl=0&p_ca=0&p_ur=IT&p_idc=Europe&nonce=jBZQ%2FyQTF%2FABtBnk&_ssign=LniS%2BhB%2BdM%2BTbkWW49k%2FlFK95ng%3D) address.
Considering, I've strictly adehered to the setup and usage guidelines, I tend to exclude errors on my side and so I do with a bug in the script itself so I believe Xiaomi has changed something, again, to prevent users from successfully unlock bootloaders.
Have a nice day! :)

Zoxorer, 2024-05-05:
is the link that appears in the browser like this?? https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi
don't wait for it to open automatically, try to open this link manually in your browser. this is actually wrong generated link

just open this link in the browser and login correctly:
https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&checkSafeAddress=true&passive=false&hidden=false

The error is in this line:
```python
if "wb_id" not in data:
    input("\nPress Enter to open confirmation page, copy link after seeing {\"R\":\"\",\"S\":\"OK\"}, and return here\n\n\033[1;33mNotice:\nIf logged in with any account in your default browser,\nplease log out before pressing Enter.\033[0m\n\n")
    conl = 'https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&checkSafeAddress=true&passive=false&hidden=false'
    if s == "Linux":
        os.system("xdg-open " + conl)
```
this can happen because the “&” symbol conflicts with the symbol in the bash syntax when the link is combined with “xdg-open” in os.system() and because the “&” symbol assumes the next thing will be run in the background

it should be like this:
`$ xdg-open 'https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&checkSafeAddress=true&passive=false&hidden=false'`
when os.system() and bash syntax execute the link with xdg-open

Guybrush3pw00d, 2024-05-06:
@Zoxorer, @offici5l 

Thanks to you both! :) Now it works perfeclty. Unfortunately, it doesn't solve my issue (and the one many others are experiencing according to what can be found around with a simple search) but this is another matter, nothing related to something wrong with MiUnlockTool but something which Xiaomi has purposedly put there to prevent users from unlocking. :(