MiForge/MiUnlockTool #39: Permission denied: 'encryptData'

kkleinw, 2024-05-24:
After entering the link, program output is as follows:

```
wb_id saved successfully
Check if the device is connected in bootloader mode...
product saved successfully
Check if the device is connected in bootloader mode...
Check if the device is connected in bootloader mode...
token saved successfully
DeviceInfo:
product: vili
token: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
AccountInfo:
id: xxxxxxxxxx
region: europe
________________________________________________________
Traceback (most recent call last):
File "/home/karl/Downloads/MiUnlockTool.py", line 222, in <module>
with open("encryptData", "wb") as edfile:
^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: 'encryptData'
```
(Account data and token are obviously different). There seems to be a problem with opening the code file? I can't quite track down the problem.

offici5l, 2024-05-24:
It looks like you are using an old version of MiUnlockTool !!
Use the latest version  1.5.1

kkleinw, 2024-05-24:
Ah, that's right. now the error is 

```
Traceback (most recent call last):
File "/home/karl/Downloads/MiUnlockTool.py", line 349, in <module>
with open(path, "wb") as edfile:
^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: 'encryptData'
```
- so, it's still the same

offici5l, 2024-05-24:
Just delete the old encryptData file

kkleinw, 2024-05-24:
I'm a big dumbdumb, issue fixed. Warning to all other users: never run this as root, it will leave files behind!

closed 2024-05-24 by kkleinw