MiForge/MiUnlockTool #40: Code 20041 MI id is not associated with a phone number, but I can't add any

mwauer, 2024-05-28:
Probably not a bug of the tool, but I'm stuck and don't know how to proceed.
I got this code when trying to unlock. Unfortunately I can't add a recovery phone to my MI account. The options suggested in https://account.xiaomi.com/helpcenter/faq/en_US/03.account-safety/04.security-phone-and-email/faq-3 aren't there. I even tried creating a new MI account on another phone so it links to it, but it didn't work.
Does anybody have an idea on how to do this? Thanks in advance!

offici5l, 2024-05-28:
Most companies currently interact with whatsapp regarding OTP, so if your phone number is linked to a whatsapp account, the confirmation code is likely to be sent to the whatsapp account rather than the phone itself. 

If you encounter the same issue, try using a temporary number. you can find many websites on google that offer this service for free. try different numbers until you get a valid one.

closed 2024-05-29 by offici5l

Ashvith10, 2025-06-15:
It looks like there is: https://offici5l.github.io/articles/mi-error-codes.html. I'll check this tomorrow once the code cool-down for Mi account is over.