MiForge/MiUnlockTool #42: Failes to unlock bootloader of xiaomi 12

rubel-sh, 2024-06-07:
code: 20038
description: 该手机已被账号 221****878 通过查找手机锁定，无法解锁

translates to : This phone has been locked by account 221****878 through Find Phone and cannot be unlocked

offici5l, 2024-06-08:
If you do not have the old account ..

There is no confirmed solution yet, but I suggest it to you ( I don't know if it will work for you or not )

Activate the phone lock again via

https://us.i.mi.com/mobile/find#/

Then reactivate this when you turn on the phone..

If the method did not work

Flash official recovery ROM via mi assistant

Then try again

rubel-sh, 2024-06-08:
update

tried first solution but still same state, locked the phone and reactactivate. 

seconds one can detect the phone when it's turned on and can do backup but on recovery mode it can't detect the phone even tho on Device Manager it shows "Android ADB Interface".

closed 2024-06-11 by offici5l

rubel-sh, 2024-06-11:
Hi , sorry for late response here's the update.

I tried to use  Windows https://www.xiaomitool.com/adb and got below errors, I've also tried other recovery roms but they produce same error.

[image]

summary
 My current rom is : Miui 14.0.6.0 TSZCNSZ (Which I can't find online)
 Tried: V14.0.7.0.TLCCNXM,  Global_OS1.0.3.0.ULCMIXM, V14.0.8.0.TLCCNXM (Recovery Roms)