MiForge/MiUnlockTool #44: Feature request: suggest rebooting into other fastboot mode when cannot obtain unlock token.

vi, 2024-07-26:
I kept getting `Failed to obtain the token! Please enter it manually: ` until I noticed that the device shows "fastbootd" in blue instead of "fastboot" in orange. After entering fastboot the other way it worked.

Look like `adb reboot fastboot` is not the same as entering fastboot using button presses.

MiUnlockTool may show a helpful message that the device may be not in the expected mode when `fastboot` platform tool works, but getting the code fails.

Also the message itself may be more specific "Failed to obtain the token _from the device_" - I initially assumed it fails to get the code from Xiaomi.

offici5l, 2024-07-27:
There is a difference between fastboot mode and bootloader mode ...

to enter bootloader mode 
```
adb reboot bootloader
```

if you are in Fastboot mode (in blue).
```
fastboot reboot bootloader 
```

vi, 2024-07-27:
Yes, but the essential part is that both respond to `fastboot` platform tool, making it easy to confuse. For example, MiUnlockTool *can* obtain a model name in fastboot mode, which made me believe that I was on the right track.

The longer message can help others, not experienced in distinguishing bootloader, fastboot, recovery and Mi Assistant modes to avoid that dead end.

PR #45 (2024-07-27): More verbose messages for failed model/token obtainment commands.
Resolves #44.
commit c42631f0 (2024-07-27): More verbose messages for failed model/token obtainment commands.

Resolves #44.
closed 2024-07-27 by offici5l