MiForge/MiUnlockTool #47: Code 30002

privatedev11, 2024-08-01:
**Describe the bug**
When I attempt to unlock my device (Redmi Note 12 5G) it gives me an error that says "Code 30002. Couldn't unlock. Please go to Mi Community to apply for authorization and try again.". My account has authorization.

**Screenshots**
[image]

offici5l, 2024-08-10:
Try changing the region by command 

in windows linux macos :

```
python3 MiUnlockTool.py global
```

in termux :

```
miunlock global
```

(or india, russia, china, europe)

If you have the same problem, use
https://github.com/offici5l/MiBypassTool

privatedev11, 2024-08-10:
Thank you for your help, but in the end, I had to use the official tool because of the wait

https://github.com/offici5l/MiBypassTool

closed 2024-08-11 by offici5l