MiForge/MiUnlockTool #53: code 10000

Samplayz2007, 2024-09-05:
So am using MacOS btw and am my device is the Xiaomi Note 9 (merlin) and am getting this error

code 10000

Request parameter error

invalid product or token

i have no idea how to fix this so please help me with this, thank you!

Samplayz2007, 2024-09-05:
Yup
DeviceInfo:
unlocked: no
SoC: None
product: merlin
token: None

offici5l, 2024-09-05:
It seems that you are in the fastboot mode, not the bootloader mode

so

1
First, try the latest version because I fixed some problems a little while ago.

2
Use the following command to make sure you are in bootloader mode :
```
fastboot reboot bootloader
```

Then restart the tool and let me know if the problem is solved.

Samplayz2007, 2024-09-05:
Alright now its 
"code 20036

Please unlock 360 hours later. And do not add your account in MIUI again, otherwise you will wait from scratch

You can unlock on: 2024-09-21 02:00"

but yeah i guess i have to wait

offici5l, 2024-09-05:
yes now wait 360 hours 
then restart the tool so you can unlock the bootloader.

Samplayz2007, 2024-09-05:
alrightyyy Thank you so much for replying so soon, i hope i didnt waste your time too much

closed 2024-09-05 by Samplayz2007