MiForge/MiUnlockTool #56: code 20033

alSeveR, 2024-09-14:
Your account is not authorized to unlock. Please change another.

I guess this is because I don't have the authorisation you get from the xiaomi comunity app. There is no way to bypass this restriction, is there?
I have been trying for several days to get the authorisation by requesting it at the time indicated and every day they add one more... It's really frustrating.
I'm sorry if this comment is inappropriate here and I appreciate if anyone knows any way to get something.

offici5l, 2024-09-14:
code 20033 has nothing to do with the xiaomi comunity app. the error is because the Xiaomi server noticed something wrong.. Maybe because you used a modified tool or an old version ...

So just change the account or create a new Xiaomi account.

___

If you are trying to bypass HyperOs restrictions to unlock the bootloader, you can use the tool [MibypassTool](https://github.com/offici5l/MiBypassTool)

note : the MiBypassTool only works with devices that support miui ... before
so If the device comes with a hypeross from the factory, the restrictions cannot be bypassed.

alSeveR, 2024-09-14:
I used another account and it worked perfectly.
Thank you very much for your tool and help.
Cheers.

closed 2024-09-14 by offici5l