MiForge/MiUnlockTool #58: Successful unlocks

badrihippo, 2024-10-01:
Since Xiaomi keeps updating the unlock procedure with small changes, I thought it would be useful to have a record of when the tool was last used successfully. If the maintainers are fine with it, we could use this issue for that? It would complement the bug reports, which only describe when the tool fails :slightly_smiling_face:

**To set the ball rolling:** I successfully used MiUnlockTool today, 2024-10-01, to unlock my **Poco F1** (MIUI) from **India** using **Linux**. It was a secondhand one and I had already created my Mi account about three weeks ago (more than 7 days basically) so the unlock happened immediately. I had used mobile data for the "Mi Unlock Status" thing but when I actually ran the tool the mobile data was switched off (I don't get mobile data signal in the house; have to go out to use it).

offici5l, 2024-10-02:
I can add a step in case of failure or success, send a message to the Telegram channel

 ex :

[image]

This will also provide statistics on the number of devices that were unlocked each year, for example...

But I have to find a way to hide the bot token so that some people don't tamper with the statistics...

offici5l, 2024-10-02:
code ex :

```
# ...

def sendunlockresult(product, unlockresult):
    try:
        requests.post('https://api.telegram.org/bot"bot-token-her"/sendMessage', data={'chat_id': 'chat-id_channel-her', 'text': f"product: {product}\nresult: {unlockresult}\ndatetime(UTC): {datetime.datetime.utcnow().strftime('%Y-%m-%d %H:%M:%S')}"})
    except:
        pass

# ...

    print(f"\nUnlock successful\n")
    sendunlockresult(product, "bootloader unlocked successfully")
except subprocess.CalledProcessError as e:
    print("Error message:", e.stderr)
    sendunlockresult(product, e.stderr)

```

badrihippo, 2024-10-05:
That's a good idea; it'd be automatic in that case and the user doesn't need to do anything! But exposing the token is a problem. One option could be to have an intermediate custom API that the bot calls, and let that API do the filtering (eg. sanity check like making sure the same account doesn't post again within 29 days, and make sure people don't send other kinds of random messages).

But that would require hosting the API somewhere; not sure what kind of resources you have. Depending on the implementation it could be done as a CGI script on a free webhost, since it's only a simple script.

Perhaps you can also extract the info from the channel and put it as a badge on the README? That would help casual users to get an idea even before joining the channel. (In my case, I don't have Telegram, so I'd have to install that and make an account before I can come and check the channel for updates)

CyberAustin, 2024-10-25:
Worked for me just now. Redmi Note 12 4G (Model: 23021RAA2Y), Global version, MIUI 14.0.10 (TMGEUCM), Android version 13 TKQ1.221114.001, Kernel 5.15.78-android13-8-00009-DIVAR-1.30410.2.32398.4

MarkusMaal, 2025-03-02:
Still worked in March 2025 with Xiaomi Pad 5 running HyperOS 1.x (with latest updates) and using the HyperSploit bypass. After waiting 15 days, that is (I was checking that the counter was going down a couple times and didn't fully turn off the device during this time). When the unlock succeeded, a factory reset was performed and FRP was tripped, meaning I had to provide my Mi account password in order to setup the device again.

tomas730, 2025-03-16:
On March 16, 2025 at 23:40 GMT+2, I unlocked my Xiaomi Redmi Note 12 Pro. I inserted a SIM card, activated Find My Devices, and immediately added my Xiaomi account to the developer options. After that, I booted the phone into fastboot mode (by pressing the power and volume down buttons) and ran the tool once. About ten days later, without turning the phone off (it was still in normal use with the SIM card inserted), I ran the tool again on the same computer, which worked and unlocked the bootloader without community app.

s0menickname, 2025-04-05:
I just successfully unlocked my Redmi 4X (santoni) with this tool, after many unsuccessful attempts with various versions of the official Mi Unlock and XiaoMiTool V2 (including from jk2K). 

Some notes (mostly for people using search engines who might find this):
1. My account had both email and phone number in it. If you register for an account and struggle to find the option to add a phone number, go to https://i.mi.com/, log in with your account, then press 'recordings.' This will prompt you to add a phone number.
2. I had issues with an SMS code not arriving while using Mi Unlock and XMT v2. Sometimes the code just doesn't go through, sometimes it's very delayed, and sometimes it's fine. 
3. **Miraculously, when I was using this tool, it allowed me to choose whether I would like to get a code by SMS _or_ email. I've never seen this behaviour with Mi Unlock or XMT v2, they've always asked me for the SMS code.** I picked the email option and sure enough, the code arrived instantly and worked.
4. With this tool, on the first attempt it gave me error code 20045. ("Please use the common user tool on the official website.
") 
I was able to bypass it by running `python3 MiUnlockTool.py global`, as advised in the guide to this project.[1]
5. To link the account to your phone, you need to: - remove all Mi accounts from your device, if there are any present, 
- insert a sim card. turn on mobile data, turn off wifi
- go to developer settings, select Mi Unlock, press a button to add account, use your credentials
- wait until it logs in and then wait a few seconds until you see a 'device successfully linked' popup
6. Xiaomi rate limits you on the number of codes you can receive on one phone number per day. They say the limit is 5 in the Mainland China and 3 elsewhere; in practice I think it's a bit more but I can't name the correct number. In any case, I had to try 3 phone numbers while messing around with Mi Unlock and XMT v2.
7. Xiaomi rate limits you on the number of accounts you can link to one phone using the same SIM. I think it's about 3, but I'm not sure. They only tell you after the fact, which is very frustrating because sometimes you've gotta make a new account for things to work.
8. On official Mi Unlock, I've had an issue when it would log me into the account, start unlocking, then the progress goes up to 99% and ends with `Can't get info, connect again`. The log showed the following error:
```
DEBUG  <18:57:03.468,T:10832> :  oem get_token -s [redacted device code] 
DEBUG  <18:57:03.505,T:10832> : FAILED (remote: 'unknown command') 
fastboot: error: Command failed 
INFO   <18:57:03.505,T:10832> : FAILED (remote: 'unknown command') 
fastboot: error: Command failed 
WARNING<18:57:03.505,T:10832> : Cannot get token.
```
So it runs `fastboot oem get_token -s [device code]` but that's not a valid command. However, `fastboot getvar token -s [device code]` works to obtain the device token, and this is how it's done in this project. The token then needs to be signed by the server[2], which this app does successfully. So if you come to this page from googling the error above, try using this app instead of Mi Unlock.
9. The whole process is designed by Xiaomi in a way that quickly makes you want to cry and smash your phone in pieces. **Don't give up.**

@offici5l, many thanks for your tool. You are a life saver.

References:
[1] Xiaomi Unlock Tool Error Codes - https://offici5l.github.io/code
[2] Xiaomi-bootloader. Some research about the bootloader used in Xiaomi phones - https://github.com/lrh2000/Xiaomi-bootloader

Colani1200, 2025-08-15:
I successfully unlocked a Poco X3 Pro (vayu) today. Because I used Xubuntu 24.04 as unpriviledged user, I needed to set up a python virtual environment and modify MiUnlockTool.py to make it use the venv accordingly. Worked like a charm, thanks @offici5l !

vsnaichuk, 2025-12-14:
I successfully unlocked the Redmi Note 7. The first attempt on macOS using this Python tool was unsuccessful, and I received the error: “description: Sorry, your MI ID is not associated with a phone number.”

I was then able to add a phone number via the Mi Flash Unlock Tool 7.6 in a Parallels virtual machine. After logging in, it suggested adding a phone number. Once the phone number was added, my second attempt with this Python tool worked flawlessly

Windoofs, 2025-12-23:
Today I tried to unlock my POCO F4 with this tool:
```
code: 20031
description: Please add your account in Settings > Developer options > Mi Unlock status.
```
The account is already bound to the phone. So I assume the Xiaomi servers are not happy with my request?

offici5l, 2025-12-23:
@Windoofs 
___

regionConfig: 

(type 'm' and press Enter to select it manually)

badrihippo, 2026-02-13:
@01GOD this tool can't bypass the waiting period. You still have to wait for the specified period before unlocking. Sometimes people don't have to wait (for example, I could unlock my second-hand phone immediately without waiting) but it is totally up to Xiaomi

midik, 2026-02-14:
just worked for Redmi 8, thanks!

murlakatamenka, 2026-06-08:
Unlocked Poco F3 with this tool:

```sh
uvx miunlock

# with parallel rebooting to FASTBOOT and not FASTBOOTD

adb reboot bootloader
```

Arch Linux, `adb` from `android-tools` and [`uv`](https://docs.astral.sh/uv/guides/tools) (to not deal with nasty Python's stuff like dependencies, virtualenvs) from official repos.

---

My initial attempts weren't successful, because 

a) my very old Mi account's region doesn't match phone's firmware, so I was getting error [20045](https://offici5l.github.io/post/miunlock-error-codes/#20045)

and

b) I was rebooting to fastboot**D**, (`adb reboot fastboot`), not fastboot (`adb reboot bootloader`). Those aren't the same ¯\\\_(ツ)_/¯

After using friend's Mi account with the matching region everything worked as expected, took me a few minutes.

---

Thanks for the software, helped me to avoid the madness of spam-clicking "approve my unlock request" in Xiaomi Community app at 00:00 @ Beijing time or using the official Windows-only Mi Unlock app.

ycftbwc, 2026-09-04:
> Unlocked Redmi 15 (creek), no problem. Using Cachyos. Perfect.

Still works, same combination

drzraf, 2026-09-29:
Redmi 7.

The real pain was getting the device "associated" with the account.
Spoiler: It'd never appeared inside "Mi account > Devices" which was empty.

But at some point I got the "unlock bootloader" phone menu to display _"Device now associated with the account"_ and even though it didn't appear in the _"Mi account > Devices"_ page, I rebooted & tried `miunlock` immediately and this time it just worked.

lfenechr, 2026-10-03:
Successful unlock (a week or two ago now) with `uvx miunlock` on CachyOS, on a Xiaomi 15 (dada) Global running HyperOS 3.0.304.

I was even blessed with a success on the first try when binding the account; I got the usual "quota reached" in Mi Community but I was able to register it in Settings and had the countdown started when checking in MiUnlock.

Unfayd, 2026-10-10:
Unlocked Succesfully On my Joyeuse Redmi Note 9 pro in Macos sequioa.  and i even i installed a custom rom on it :> Thanks Again

[image]

Darius499, 2026-10-12:
Successfully unlocked Xiaomi Mi 10T Pro (apollon) running MIUI 14.0.1. using Linux Mint 21.3.

Until I removed installed adb and fastboot, it was stuck on sending encryptData. But after tool downloaded platform-tools, it worked. Unlocked without problem.

Thank you offici5l for this tool.

JackNeedHelp, 2026-10-14:
how to  unlock poco f1 bootloader, bypass bootloader, Mi account, without Bind your account

hubertwalter05-eng, 2026-10-22:
J'ai pas réussi la mienne.. Redmi note 9 Pro 5g gauguin....est que quelqu'un peut m'aider ?

fluffy-the-cat, 2026-10-24:
Successful unlock on Redmi Note 9S