MiForge/MiUnlockTool #62: How to fix code20045

Ligma123-coder, 2024-10-11:
Please use common user tool on the official website

akanouras, 2024-10-18:
From https://github.com/RohitVerma882/termux-miunlock/issues/73#issuecomment-2295278915 :

> Your region is wrong, try another one. I had the same issue when giving the script that gets the key the argument global, I tried with europe and it worked. Just try to run the script will all regions (india, global, china, russia, europe) and see if it works.

For example: `python3 MiUnlockTool.py global` or `python3 MiUnlockTool.py europe`.

LS-98nox, 2024-12-03:
Tengo autorización de desbloqueo y me lanza error 20045 dice : error 20045 la región no coincide con el lugar donde se vende el dispositivo.

offici5l, 2024-12-04:
On October 20, 2024, Xiaomi updated its bootloader unlocking policy for Chinese devices.

If you own a Chinese device and try to unlock the bootloader using a global Xiaomi account, you will encounter the following error:

code 20045 = The place where the account is registered does not match the place where the phone is sold.

This means it is not possible to unlock the bootloader of a Chinese device using a global Xiaomi account.

Solution:

Requirements: Xiaomi Chinese account (Xiaomi account registered in the Chinese region)

Steps:
- Link the account xiaomi in the developer options.

- If you are using Termux, execute the command:
```
miunlock china
```
- For other systems, use the command:
```
python3 Miunlocktool.py china
```

offici5l, 2025-04-06:
Bootloader Unlock Restriction: Code 20045 = "The place where the account is registered does not match the place where the phone is sold." Solution:

offici5l.github.io/Articles/code_20045

micw, 2025-06-15:
I had 2 of these devices here. One I could unlock easily with my german Xiaomi account (europe or global variant). For other I needed an additional phone number and I had to register via Chineese account. Here's the documentation how to do so:

https://web.archive.org/web/20250319034506/https://offici5l.github.io/Articles/code_20045

Creating a chineese account with german number worked without any problems.

closed 2025-12-14 by offici5l

hubertwalter05-eng, 2026-10-25:
Solution qui marche en 2026 .. s'il vous plaît ?