MiForge/MiUnlockTool #66: the script shuts without displaying error

donttoutch, 2024-11-05:
Heloo when i run the script it stops at this line : 
`print(f"\n{cg}DeviceInfo:{cres}\nunlocked: {unlocked}\nSoC: {SoC}\nproduct: {product}\ntoken: {token}\n")
`
and i actually get this code at the end : 
```
DeviceInfo:
unlocked: no
SoC: Qualcomm
product: aurora
token: xxxx (with my actual token)
```

and shuts 
is it the decrypted token? what should i do after that then?

thanks in advance!!

donttoutch, 2024-11-05:
ok so by adding "europe" it get one step further to the class RetrieveEncryptData 

and i get this error : 

```
Traceback (most recent call last):
  File "C:\Users\amine\Downloads\MiUnlockTool.py", line 337, in <module>
    c = RetrieveEncryptData("/api/v2/unlock/device/clear", {"data":{"product":product}}).add_nonce().run()
  File "C:\Users\amine\Downloads\MiUnlockTool.py", line 318, in add_nonce
    r = RetrieveEncryptData("/api/v2/nonce", {"r":''.join(random.choices(list("abcdefghijklmnopqrstuvwxyz"), k=16)), "sid":"miui_unlocktool_client"}).run()
  File "C:\Users\amine\Downloads\MiUnlockTool.py", line 333, in run
    return json.loads(b64decode((lambda s: s[:-s[-1]])(AES.new(b64decode(ssecurity), AES.MODE_CBC, b"0102030405060708").decrypt(b64decode(session.post(Url(scheme="https", host=url, path=self.path).url, data=self.params, headers=headers, cookies=cookies).text)))))
  File "C:\Users\amine\Downloads\MiUnlockTool.py", line 333, in <lambda>
    return json.loads(b64decode((lambda s: s[:-s[-1]])(AES.new(b64decode(ssecurity), AES.MODE_CBC, b"0102030405060708").decrypt(b64decode(session.post(Url(scheme="https", host=url, path=self.path).url, data=self.params, headers=headers, cookies=cookies).text)))))
IndexError: index out of range
```

can you help please?

donttoutch, 2024-11-05:
ok so by adding "europe" it get one step further to the class RetrieveEncryptData

and i get this error :

```
Traceback (most recent call last):
  File "C:\Users\x\Downloads\MiUnlockTool.py", line 337, in <module>
    c = RetrieveEncryptData("/api/v2/unlock/device/clear", {"data":{"product":product}}).add_nonce().run()
  File "C:\Users\x\Downloads\MiUnlockTool.py", line 318, in add_nonce
    r = RetrieveEncryptData("/api/v2/nonce", {"r":''.join(random.choices(list("abcdefghijklmnopqrstuvwxyz"), k=16)), "sid":"miui_unlocktool_client"}).run()
  File "C:\Users\x\Downloads\MiUnlockTool.py", line 333, in run
    return json.loads(b64decode((lambda s: s[:-s[-1]])(AES.new(b64decode(ssecurity), AES.MODE_CBC, b"0102030405060708").decrypt(b64decode(session.post(Url(scheme="https", host=url, path=self.path).url, data=self.params, headers=headers, cookies=cookies).text)))))
  File "C:\Users\x\Downloads\MiUnlockTool.py", line 333, in <lambda>
    return json.loads(b64decode((lambda s: s[:-s[-1]])(AES.new(b64decode(ssecurity), AES.MODE_CBC, b"0102030405060708").decrypt(b64decode(session.post(Url(scheme="https", host=url, path=self.path).url, data=self.params, headers=headers, cookies=cookies).text)))))
IndexError: index out of range
```
can you help please? i've tried multiple times

offici5l, 2024-11-05:
Try this to print more info to know the cause of the issue

https://github.com/offici5l/MiUnlockTool/blob/dev/MiUnlockTool.py

donttoutch, 2024-11-05:
hello! 
thank you! so i have more information i added some debug to the script : 

`
url = https://eu-unlock.update.intl.miui.com/api/v2/nonce
Error: Received non-200 status code 401
<function urlunparse at 0x0000017045003E20>
Traceback (most recent call last):
  File "C:\Users\amine\MiUnlock\pythonProject\MiUnlockTool.py", line 425, in <module>
    c = RetrieveEncryptData("/api/v2/unlock/device/clear", {"data":{"product":product}}).add_nonce().run()
  File "C:\Users\amine\MiUnlock\pythonProject\MiUnlockTool.py", line 323, in add_nonce
    nonce_response = self.get_nonce()
  File "C:\Users\amine\MiUnlock\pythonProject\MiUnlockTool.py", line 335, in get_nonce
    nonce_response = RetrieveEncryptData("/api/v2/nonce", nonce_params).run()
  File "C:\Users\amine\MiUnlock\pythonProject\MiUnlockTool.py", line 365, in run
    decrypted_data = self.decrypt_response(response)
  File "C:\Users\amine\MiUnlock\pythonProject\MiUnlockTool.py", line 398, in decrypt_response
    raise ValueError("Received empty response or failed to decrypt.")
ValueError: Received empty response or failed to decrypt.
`

and using your script i got this : 
`Error: index out of range

Response Text: 

Decoded Data: b''

Decrypted Data: b''

Process finished with exit code 0
`

offici5l, 2024-11-05:
download and try again :
https://github.com/offici5l/MiUnlockTool/blob/dev/MiUnlockTool.py

Also tell me more info like :
region account
your region
device region cn eu mi in ...
Have you tried logging out of your account in the browser and logging in again?

donttoutch, 2024-11-05:
region account : France - Europe
my region : France - Europe
Device is "eu" normally
how do I log out on the browser?

offici5l, 2024-11-05:
Open the link in your default browser. 
https://account.xiaomi.com
And click on log out

Then when you run MiUnlockTool, type 2 and press Enter.

donttoutch, 2024-11-05:
first i had this problem : 
if 'serviceToken' not in cookies:
    print(f"\n{cr}Failed to get serviceToken.{cres}")
    remove("wb_id")
    
so i loaded the link on my computer to enter the token manually

here's the result : 
________________________________________________________

URL: https://eu-unlock.update.intl.miui.com/api/v2/nonce

Error: index out of range

Response Text:

Decoded Data: b''

Decrypted Data: b''

donttoutch, 2024-11-05:
by the way i forgot to precise i have ah Xiaomi 14 Ultra on hyperOS

offici5l, 2024-11-05:
the issueis related to the failure to obtain nonce, the Xiaomi server did not accept the data sent for some reason, perhaps Xiaomi updates regarding modern devices ... !

Question about the official tool what error do you get?

donttoutch, 2024-11-05:
when i go in settings > unlock bootloader i get : go to Xiaomi Community first

offici5l, 2024-11-05:
I don't mean the settings app, but the official Xiaomi tool to unlock the bootloader 

( I added a step to find out exactly what the issue is.
https://github.com/offici5l/MiUnlockTool/blob/dev/MiUnlockTool.py )

donttoutch, 2024-11-06:
i was only having an error at first so i edited the script : 

`        except Exception as e:
            print("\nError:", e)
            if isinstance(response, str):
                print(f"\nResponse (as string):\n{response}")
            else:
                try:
                    print(f"\nData sent with the request: {response.request.body.decode('utf-8')}")
                except AttributeError:
                    print("\nCould not access request data")
            print("\n\nError:", e)
            if 'response_text' in locals():
                print(f"\n\nresponse_text:\n\n{response_text}\n\n")
            if 'decoded_data' in locals():
                print(f"\n\ndecoded_data:\n\n{decoded_data}\n\n")
            if 'decrypted_data' in locals():
                print(f"\n\ndecrypted_data:\n\n{decrypted_data}\n\n")
            if 'unpadded_data' in locals():
                print(f"\n\nunpadded_data:\n\n{unpadded_data}\n\n")
            if 'decoded_unpadded_data' in locals():
                print(f"\n\ndecoded_unpadded_data:\n\n{decoded_unpadded_data}\n\n")
            if 'json_data' in locals():
                print(f"\n\njson_data:\n\n{json_data}\n\n")

            exit()`
            
            and go this answer : 
            
Error: 'str' object has no attribute 'text'

Response (as string): xxxxxxxxxx _(i removed it as i'm not sure if its confidential data but i got a string with a length of 281)_

Error: 'str' object has no attribute 'text'

Process finished with exit code 0

donttoutch, 2024-11-06:
Okay! 

so i found where the problem was : 

```
        try:
            response = session.post(Url(scheme="https", host=url, path=self.path).url, data=self.params,
                                    headers=headers, cookies=cookies).text
            response_text = response**.text**
```
i removed .text
and i got this : 
```
This device clears user data when it is unlocked

An unlocked device is an easy target for malware which may damage your device or cause financial loss.

Press Enter to Unlock
( or type q and press Enter to quit)

________________________________________________________

code 20031

Please add your account in Settings > Developer options > Mi Unlock status.

offici5l.github.io/code
```

but the problem is i can't add my account in the settings, it tells  "Couldn't add. Please go to Mi Community to apply for authorization"

when i use your MiBypassTool.py to bypass (i installed your downgrade settings.apk) i get this : 

```
Check if device is connected in normal mode...

Device is connected Done

Now bind your account in Mi Unlock status ...

Current version: V816.0.13.0.UNAEUXM

Version updated to: V14.0.13.0.UNAEUXM

code: 10000

description: 参数异常

descCN: 参数异常

descEN: Request parameter error
```

closed 2024-11-06 by offici5l