MiForge/MiUnlockTool #67: I have the following problem

liumingyuer, 2024-11-06:
My phone model：mi11
area：China
System version：hyperOS 1.0.1.0
my Mac tried running MiUnlockTool.py

Traceback (most recent call last):
  File "/Users/liumingyue/Library/Mobile Documents/com~apple~CloudDocs/Downloads/MiUnlockTool.py", line 21, in <module>
    import re, requests, json, hmac, random, binascii, urllib, hashlib, io, urllib.parse, time, sys, urllib.request, zipfile, webbrowser, platform, subprocess, shutil, stat, datetime, threading
ModuleNotFoundError: No module named 'requests'

offici5l, 2024-11-06:
open your terminal
Run the following command:
```
pip install requests pycryptodomex urllib3 colorama
```

closed 2024-11-06 by offici5l

Duarte475, 2024-11-18:
I also have the same problem and installing the dependencies didn't solve it.
I have a Mi 11 Lite 5G (renoir), EEA region.
I'm running on macOS with Python 3.13.0

Duarte475, 2024-11-18:
> I also have the same problem and installing the dependencies didn't solve it. I have a Mi 11 Lite 5G (renoir), EEA region. I'm running on macOS with Python 3.13.0

I was able to fix this issue by running on a virtual environment

LeeBinder, 2025-04-06:
pip3 install requests pycryptodomex urllib3 colorama --break-system-packages