MiForge/MiUnlockTool #74: Please unlock 122 hours later.

vit-cerny, 2024-12-11:
`code 20036

Please unlock 122 hours later. And do not add your account in MIUI again, otherwise you will wait from scratch

You can unlock on: 2024-12-16 22:00

________________________________________________________

Press Enter to exit ...`

I thought this tool is made to bypass wait...

MarkusMaal, 2025-03-02:
If you have a MediaTek device, it might be possible to bypass in EDL mode, but if you have a Snapdragon device, it is impossible to bypass the MIUI unlock wait time with free tools, that is something only Xiaomi can do on their side. This tool is just an alternative to the official Mi Unlock tool that works cross platform. I know it's annoying, but if you are able to get this message to pop up, it means you'll most likely be able to unlock once the time runs out.

LeeBinder, 2025-03-25:
I'm also confronted with the dreaded 168 hrs waiting time on my POCO X3 Pro (Qualcomm Snapdragon). I read somewhere "The waiting period may vary from person to person". Due to partial malfunction of the new SIM my provider had sent me, mobile data kept flaking out, resulting in issues during the various login-in and verification processes, which maybe is the reason I "scrored" this long waiting period?

Those who already manage to get your bootloader unlocked: how long did YOU actually have to wait?? Or did the unlock procedure even run through for some of you right away w/o any delay?

I'm wondering, has anyone tried to bypass the waiting time by:
1. extract the existing SIM and do a factory reset
2. insert a different SIM card with a new mobile phone #
3. sign into MIUI with a different gmail address 
4. start the unlock process fresh

or do the Xiaomi unlock servers which manage the wating time, ONLY determine the phone by its hardware, regardless of SIM and account email address?

Would very much appreciate educted insights 👍- thanks.

LeeBinder, 2025-04-17:
@Anmol-devOP no, I had way too much other stuff to tackle, so I didn't get a chance once the 122 hrs had passed, then successfully unlocked the bootloader with the MiUnlockTool in macOS 👍

tymur999, 2025-05-21:
> I'm also confronted with the dreaded 168 hrs waiting time on my POCO X3 Pro (Qualcomm Snapdragon). I read somewhere "The waiting period may vary from person to person". Due to partial malfunction of [...]

SIM is necessary only in starting the wait and then unlocking. As long as you don't factory reset, you should be fine. No logging out as well.

theangryhat7892, 2025-09-06:
> [@Anmol-devOP](https://github.com/Anmol-devOP) no, I had way too much other stuff to tackle, so I didn't get a chance once the 168 hrs had passed, then successfully unlocked the bootloader with the MiUnlockTool in macOS 👍

could i still use the phone during the waiting period

closed 2025-12-14 by offici5l