MiForge/MiUnlockTool #85: < waiting for any device >

AgentSprout228, 2025-02-11:
The python file is stuck on < waiting for any device > exen considering im in fastboot mode, what should i do?

nezmar48, 2025-02-20:
I have the same issue on Linux

pure-eAIa, 2025-04-07:
@AgentSprout228 @nezmar48 https://developer.android.com/studio/run/win-usb

y0shir, 2025-04-11:
> [@AgentSprout228](https://github.com/AgentSprout228) [@nezmar48](https://github.com/nezmar48) https://developer.android.com/studio/run/win-usb

This only applies to windows. I have installed adbd and fastboot from bookworm-backports and I still stuck at  < waiting for any device >

kumarvivek1752, 2025-05-01:
i've same issue and usb driver already installed how to verify?? 

[image]