MiForge/MiUnlockTool #89: Thank you!

tomas730, 2025-03-16:
On March 16, 2025 at 23:40 GMT+2, I unlocked my Xiaomi Redmi Note 12 Pro. I inserted a SIM card, activated Find My Devices, and immediately added my Xiaomi account to the developer options. After that, I booted the phone into fastboot mode (by pressing the power and volume down buttons) and ran the tool once. About ten days later, without turning the phone off (it was still in normal use with the SIM card inserted), I ran the tool again on the same computer, which worked and unlocked the bootloader without community app.

closed 2025-03-16 by tomas730

juancfigueroa, 2025-04-25:
Hi Tomas. Did unlocking the bootloader on your Xiaomi Redmi Note 12 Pro cause a wipe of your device?

tomas730, 2025-04-25:
Yes, but it did warn me before wiping storage in cmd.